wendehals/arduino_sct_tools #1: Cannot upload to Arduino

victorrsoliveira, 2016-10-02:
I've tried to use your tool from the example Blink, followed all steps that you just provided in README page but wasn't very succesfull. This is the error I am getting from Eclipse console:

```
avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/victor/.arduinocdt/packages/arduino/tools/avrdude/6.3.0-arduino6/etc/avrdude.conf"
         User configuration file is "/home/victor/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : -b115200
         Using Programmer              : arduino
avrdude: ser_open(): can't open device "-b115200": No such file or directory

avrdude done.  Thank you.
```
I've done exactly every single step you've just described. Arduino IDE is already installed on my machine and I am able to program and upload code to my Arduino by using it.
Do you think of anything that might be missing in my  toolchain?

wendehals, 2016-11-24:
Sorry, I didn't work on the project for quite a while. Have you managed to solve your problem? I guess it's a problem with your serial connection to your Arduino. Have you tried to program your Arduinio from within the original Arduino IDE? Did you create a Launch Target under the Connections view in the C/C++ perspective?

wendehals, 2016-11-24:
It is not an issue related to the arduino_sct_tools. It a problem related to configuration of the AVR tool chain. So I will close this issue.

closed 2016-11-24 by wendehals