wendehals/arduino_sct_tools #7: 'java.lang.NullPointerException' when trying to build code artifacts with YAKINDU SCT 3.0

PeronistaUniversal, 2017-07-24:
Using YAKINDU SCT 3.0 + Arduino SCT Tools, I get this error when trying to build code artifacts:

```
'Execute SCT Genmodel test.gen' has encountered a problem.
java.lang.NullPointerException
```

Also, the directory 'src-gen' is not automatically created. This doesn't happen using an older version or YAKINDU SCT, or when creating a YAKINDU project without Arduino tools.

brunolinardi, 2017-09-28:
I'm also having the exact same issue. Trying to make it work, but I get stuck in this error.

The console gives me the following errors:

`Building Blink
"C:/Users/Bruno/.arduinocdt/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "./Blink.elf"   "./core.a" "-L." -lm
C:\Users\Bruno\AppData\Local\Temp\ccyAbIWo.ltrans0.ltrans.o: In function `main':
C:/Users/Bruno/.arduinocdt/packages/arduino/hardware/avr/1.6.20/cores/arduino/main.cpp:43: undefined reference to `setup'
C:/Users/Bruno/.arduinocdt/packages/arduino/hardware/avr/1.6.20/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
makefile:54: recipe for target 'Blink.elf' failed
make: *** [Blink.elf] Error 1`

Could someone please help?

Thanks!

wendehals, 2017-10-23:
There is a new release 0.7.0 for YAKINDU SCT 3.1.0 which fixes the issue

closed 2017-10-23 by wendehals

PeronistaUniversal, 2017-11-03:
Yakindu SCT version 3.1.1
YAKINDU Statechart Tools for Arduino	0.7.0.201710232100

I'm still getting the same error:

Building 1234
"C:/Users/Carlos2/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\Users\Carlos2\.arduinocdt\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lib" "-LC:\Users\Carlos2\.arduinocdt\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "./1234.elf" -Wl,--start-group   "./arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-L."
./arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
./arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
./arduino.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':
C:/Users/Carlos2/.arduinocdt/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:110: undefined reference to `setup'
makefile:78: recipe for target '1234.elf' failed
C:/Users/Carlos2/.arduinocdt/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:110: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
make: *** [1234.elf] Error 1

wendehals, 2017-11-05:
There is actually an incompatibility to Yakindu SCT 3.1.1. A new release is in preparation and will be released today.

PeronistaUniversal, 2017-11-10:
I updated to version 0.7.1 and still the same error appears when builiding:

 YAKINDU Statechart Tools for Arduino	0.7.1.201711051429	org.yakindu.sct.arduino.feature.group	Lothar Wendehals

  Aeri-feature	3.1.1.201710271511	com.yakindu.sct.aeri.feature.feature.group	itemis AG
  C/C++ Development Tools	9.3.3.201710130142	org.eclipse.cdt.feature.group	Eclipse CDT
  Code Recommenders News Feed	2.4.10.v20170911-1410	org.eclipse.recommenders.news.rcp.feature.feature.group	Eclipse Code Recommenders
  Eclipse Platform	4.7.1.v20171009-0410	org.eclipse.platform.feature.group	Eclipse.org
  Eclipse Plug-in Development Environment	3.13.1.v20171009-0537	org.eclipse.pde.feature.group	Eclipse.org
  Xtend IDE	2.12.0.v20170519-1412	org.eclipse.xtend.sdk.feature.group	Eclipse Xtend
  YAKINDU Headless Code Generator	3.1.1.201711081247	com.yakindu.sct.generator.application.feature.feature.group	itemis AG
  YAKINDU License Feature	3.1.1.201711081247	com.yakindu.sct.license.feature.feature.group	itemis AG
  YAKINDU License Integration For Standard Edition	3.1.1.201711081247	com.yakindu.sct.license.standard.feature.feature.group	itemis AG
  YAKINDU Products Feature License	2.0.0.201708011008	com.yakindu.products.license.feature.feature.group	itemis AG
  YAKINDU SCTUnit (BETA)	3.1.1.201711081247	com.yakindu.sctunit.feature.feature.group	itemis AG
  YAKINDU SCTUnit C Generator	3.1.1.201711081247	com.yakindu.sctunit.generator.c.feature.feature.group	itemis AG
  YAKINDU SCTUnit CPP Generator	3.1.1.201711081247	com.yakindu.sctunit.generator.cpp.feature.feature.group	itemis AG
  YAKINDU SCTUnit Headless Generator Support	3.1.1.201711081247	com.yakindu.sctunit.generator.application.feature.feature.group	itemis AG
  YAKINDU SCTUnit Java Generator	3.1.1.201711081247	com.yakindu.sctunit.generator.java.feature.feature.group	itemis AG
  YAKINDU Statechart Tools	3.1.1.201710271438	org.yakindu.sct.feature.group	statecharts.org
  YAKINDU Statechart Tools - Product	3.1.1.201710271511	com.yakindu.sct.products.feature.feature.group	itemis AG
  YAKINDU Statechart Tools C Code Generator	3.1.1.201710271438	org.yakindu.sct.generator.c.feature.group	statecharts.org
  YAKINDU Statechart Tools C++ Code Generator	3.1.1.201710271438	org.yakindu.sct.generator.cpp.feature.group	statecharts.org
  YAKINDU Statechart Tools for Arduino	0.7.1.201711051429	org.yakindu.sct.arduino.feature.group	Lothar Wendehals
  YAKINDU Statechart Tools Java Code Generator	3.1.1.201710271438	org.yakindu.sct.generator.java.feature.group	statecharts.org
  YAKINDU Statechart Tools Standard Edition	3.1.1.201710271511	sct	null

> "C:/Users/Carlos2/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\Users\Carlos2\.arduinocdt\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lib" "-LC:\Users\Carlos2\.arduinocdt\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "./TestDos.elf" -Wl,--start-group   "./arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-L."
./arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
./arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
./arduino.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':
C:/Users/Carlos2/.arduinocdt/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:110: undefined reference to `setup'
C:/Users/Carlos2/.arduinocdt/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:110: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
make: *** [TestDos.elf] Error 1
makefile:78: recipe for target 'TestDos.elf' failed

wendehals, 2018-05-09:
Do you still encounter the problem? Try the newest release 0.8.0 with YAKINDU SCT 3.3.2. If it is still there, could you please share the complete stack trace/error output printed after the java.lang.NullPointerException?