wendehals/arduino_sct_tools #11: Errors with Yakindu SCT 3.2.0

PeronistaUniversal, 2018-01-12:
Trying using the code generator with Yakindu SCT 3.2.0 returns this error:

Guice creation errors:

> 1) No implementation for org.yakindu.sct.generator.c.IncludeProvider was bound.
>   while locating org.yakindu.sct.generator.c.IncludeProvider
>     for field at org.yakindu.sct.generator.c.StatemachineHeader._includeProvider(Unknown Source)
>   while locating org.yakindu.sct.generator.cpp.StatemachineHeader
>     for field at org.yakindu.sct.arduino.generator.cpp.ArduinoCodeGenerator.statemachineHeaderContent(Unknown Source)
>   at org.yakindu.sct.arduino.generator.cpp.ArduinoCppCodeGeneratorModule.configure(ArduinoCppCodeGeneratorModule.java:36)
> 
> 1 error
> Guice creation errors:
> 
> 1) No implementation for org.yakindu.sct.generator.c.IncludeProvider was bound.
>   while locating org.yakindu.sct.generator.c.IncludeProvider
>     for field at org.yakindu.sct.generator.c.StatemachineHeader._includeProvider(Unknown Source)
>   while locating org.yakindu.sct.generator.cpp.StatemachineHeader
>     for field at org.yakindu.sct.arduino.generator.cpp.ArduinoCodeGenerator.statemachineHeaderContent(Unknown Source)
>   at org.yakindu.sct.arduino.generator.cpp.ArduinoCppCodeGeneratorModule.configure(ArduinoCppCodeGeneratorModule.java:36)
> 
> 1 error

Zarten, 2018-01-24:
same problem here

Bukethead, 2018-02-25:
I'm having this problem as well.

wendehals, 2018-05-07:
Problem fixed with release 0.8.0 and YAKINDU SCT 3.3.2

closed 2018-05-07 by wendehals